vlachoudis/bCNC #1824: Manual Tool Change (TLO) doesn't work

SamsonBox, 2023-03-18:
Hi,
I tried the manual tool change (TLO) and I found out, that it doesn't work. In fact all the motions and calculation are right, but the TOL doesn't survive the `purgeController`  at all. The reason is the folloing:
After all motions are done, bCNC state is changed to `Idle` in the `controllerStateChange` function and `jobDone` is called which in turn calls `purgeController`  from the `serialIO`-Thread. The `purgeController` stores the  "G"-variables and the "TLO", resets the controller (via a realtime commad) a and tries restores the "G"-variables and the "TLO". But in the next run `serialIO`-loop the reset answer ("Grbl xxx") ist parsed and the `_stop` is set to `True`. Due to this flag the gcode-queue gets cleared (`emptyQueue()`) and the commands to restore the "G"-variables and the "TLO" are gone. Have I done something wrong? Or do you have any suggestions to overcome ths problem?

Thank you

rschell, 2023-03-19:
I have noticed these same issues over the past year.  There are a number of fixes IMO that need to be applied to get TLO and Tools to function as expected. You might give these changes try, [fix_TLO_and_Tool_handling](https://github.com/rschell/bCNC/tree/fix_TLO_and_Tool_handling)

MARIOBASZ, 2023-03-20:
I am going to try. I have always used M6 with TLO and no problems.

SamsonBox, 2023-03-20:
@rschell I have quickly looked over your changes. When I understand your proposel correctly, than you store the values in a file and reload them after the controller reset. I'm sure if that ist the way to go. I think the controller reset should wait until the reset is really over and send the commands after wards.

rschell, 2023-03-20:
My computer/controller on my machine spend 98% of its time in a development mode.  The operating system/bCNC/controller are rebooted quite often.  The state of the machine doesn't change during these reboots.  So my goal was to return bCNC/machine state to that prior to these resets.

The code snippets you maybe most interested in are in _GenericController.py (softReset and purgeController).  I spent a few days figuring out that g43.1 in the G return only indicates that a value has been set and is not sufficient to reset the state.

SamsonBox, 2023-03-20:
You are right, these changes are good. But I think the soft_reset should become an event to get it dispatched from the serialIO thread and wait till the reset-reply is received

rschell, 2023-03-20:
I'll take a look at your suggestion.

JipeB, 2023-03-22:
Is it necessary to purge the controller at the end of the program?

In IMD mode you can chain commands without losing the TLO