vlachoudis/bCNC #1832: Involute gear, internal gear

MARIOBASZ, 2023-04-13:
Great job. But you have to correct the shape in the internal gear: what points to the center must point to the outside, what points to the outside must point to the center (it looks like a very nice balloon for children anyway).

MARIOBASZ, 2023-04-13:
[image]

MARIOBASZ, 2023-04-13:
It should be more like this.
Adding gear rack will be very useful. For my personal use, I had modified gear for internal gear and for rack. Mediocre result, very enthusiastic about your version. Hugs.

MARIOBASZ, 2023-04-13:
[image]

LittlePierre, 2023-04-13:
_"It should be more like this.
Adding gear rack will be very useful. For my personal use, I had modified gear for internal gear and for rack. Mediocre result, very enthusiastic about your version. Hugs."_

Sorry @MARIOBASZ  but I think that the profile you describe is wrong.

The profile generated by the plugin is good. You just have to add a circle around the internal gear to get the rack, depending your application.

Pierre

[image]
[image]

LittlePierre, 2023-04-16:
_you have to correct the shape in the internal gear: what points to the center must point to the outside, what points to the outside must point to the center_

Noooooo ! :-)

here is how it works : 

[image]

LittlePierre, 2023-04-16:
In red is the profile generated by involutegear plugin.
You can see that it matches the theoretical gear.
Within the plugin you can 
* either let the default parameters, corresponding to ISO Rack specifications,
* either set custom parameters, such as customized addendum, deddendum, or fillet coeff.

[image]

MARIOBASZ, 2023-04-18:
It may be that I am wrong.
My concern is the following: the tooth has a trapezoidal shape, and a rounding / recess in the head of both teeth to allow the rotation of the gears. It would seem that the recess is being fulfilled at the base, instead of at the head. It may be me who is wrong. However, it would seem that they have done so in this image.
[image]

LittlePierre, 2023-04-18:
There are only two profiles for gears : 
- the standard one for most applications is involute. It is generated by the involute plugin of bCNC
- some rare applications such as watchmaking use the cycloid specific profile - not covered (yet)

Believe a man who had spent 30 years of his carrier in french capital of watchmaking industry.

LittlePierre, 2023-04-18:
https://en.m.wikipedia.org/wiki/Involute_gear

LittlePierre, 2023-04-18:
You can play with addendum, dedendum, shift coeffs to change the shape of the tooth, but be sure you understand what you're doing. 
Otherwise use default parameters which are the normalized ones.

LittlePierre, 2023-04-19:
>  I don't know if your inner gear design allows your partner to roll. 

Yes it does.
Sorry to say so Mario, and I say it with respect, but you are trying to reinvent square wheel.

Pierre

MARIOBASZ, 2023-04-19:
good Pierre. I have seen images with both ways: the silhouette of the tooth pointing towards the center, like the image I showed, (the way of meshing is comparable to that of two external gears) and the silhouette of the tooth pointing outwards. You must be right. Hug.