vlachoudis/bCNC #1834: build bcnc.exe

macfifa, 2023-05-07:
i installed bcnc ver. 0.9.15 when i use the "python -m bCNC" the program works  with no error then i tried making bcnc.exe build
and followed the wiki .
pyinstaller --onefile --distpath . --hidden-import tkinter --paths lib;plugins;controllers --icon bCNC.ico --name bCNC __main__.py
The program dont work giving me error no helper module found
python 3.10.6
win 10

macfifa, 2023-05-08:
i tried deleting __init__.py and the error no helper is gone then the program works but there are errors with numpy
missing module named _frozen_importlib_external - imported by importlib._bootstrap (delayed), importlib (optional), importlib.abc (optional), zipimport (top-level)
excluded module named _frozen_importlib - imported by importlib (optional), importlib.abc (optional), zipimport (top-level)
missing module named org - imported by pickle (optional)
missing module named pwd - imported by posixpath (delayed, conditional, optional), shutil (delayed, optional), tarfile (optional), pathlib (delayed, optional), subprocess (delayed, conditional, optional), netrc (delayed, conditional), getpass (delayed), webbrowser (delayed), http.server (delayed, optional)
missing module named grp - imported by shutil (delayed, optional), tarfile (optional), pathlib (delayed, optional), subprocess (delayed, conditional, optional)
missing module named posix - imported by os (conditional, optional), posixpath (optional), shutil (conditional), importlib._bootstrap_external (conditional)
missing module named resource - imported by posix (top-level)
missing module named 'org.python' - imported by copy (optional), xml.sax (delayed, conditional)
missing module named 'java.lang' - imported by platform (delayed, optional), xml.sax._exceptions (conditional)
missing module named multiprocessing.BufferTooShort - imported by multiprocessing (top-level), multiprocessing.connection (top-level)
missing module named multiprocessing.AuthenticationError - imported by multiprocessing (top-level), multiprocessing.connection (top-level)
missing module named _posixshmem - imported by multiprocessing.resource_tracker (conditional), multiprocessing.shared_memory (conditional)
missing module named _posixsubprocess - imported by subprocess (conditional), multiprocessing.util (delayed)
missing module named multiprocessing.get_context - imported by multiprocessing (top-level), multiprocessing.pool (top-level), multiprocessing.managers (top-level), multiprocessing.sharedctypes (top-level)
missing module named multiprocessing.TimeoutError - imported by multiprocessing (top-level), multiprocessing.pool (top-level)
missing module named fcntl - imported by subprocess (optional), serial.serialposix (top-level)
missing module named multiprocessing.set_start_method - imported by multiprocessing (top-level), multiprocessing.spawn (top-level)
missing module named multiprocessing.get_start_method - imported by multiprocessing (top-level), multiprocessing.spawn (top-level)
missing module named pyimod02_importers - imported by C:\Users\HaSaN\AppData\Roaming\Python\Python311\site-packages\PyInstaller\hooks\rthooks\pyi_rth_pkgutil.py (top-level)
missing module named _scproxy - imported by urllib.request (conditional)
missing module named termios - imported by getpass (optional), serial.serialposix (top-level), tty (top-level)
missing module named 'System.IO' - imported by serial.serialcli (top-level)
missing module named System - imported by serial.serialcli (top-level)
missing module named clr - imported by serial.serialcli (top-level)
missing module named olefile - imported by PIL.FpxImagePlugin (top-level), PIL.MicImagePlugin (top-level)
missing module named packaging - imported by PIL.Image (delayed, conditional, optional)
missing module named cffi - imported by PIL.Image (optional), PIL.PyAccess (optional)
missing module named defusedxml - imported by PIL.Image (optional)
missing module named 'scipy.special' - imported by svgelements.svgelements (delayed)
missing module named scipy - imported by svgelements.svgelements (delayed)
missing module named vms_lib - imported by platform (delayed, optional)
missing module named java - imported by platform (delayed)
missing module named _winreg - imported by platform (delayed, optional)
missing module named readline - imported by cmd (delayed, conditional, optional), code (delayed, conditional, optional), pdb (delayed, optional)
missing module named psutil - imported by numpy.testing._private.utils (delayed, optional)
missing module named _dummy_thread - imported by numpy.core.arrayprint (optional)
missing module named numpy.core.result_type - imported by numpy.core (delayed), numpy.testing._private.utils (delayed)
missing module named numpy.core.float_ - imported by numpy.core (delayed), numpy.testing._private.utils (delayed)
missing module named numpy.core.number - imported by numpy.core (delayed), numpy.testing._private.utils (delayed)
missing module named numpy.core.object_ - imported by numpy.core (top-level), numpy.linalg.linalg (top-level), numpy.testing._private.utils (delayed)
missing module named numpy.core.all - imported by numpy.core (top-level), numpy.linalg.linalg (top-level), numpy.testing._private.utils (delayed)
missing module named numpy.core.bool_ - imported by numpy.core (delayed), numpy.testing._private.utils (delayed)
missing module named numpy.core.inf - imported by numpy.core (delayed), numpy.testing._private.utils (delayed)
missing module named numpy.core.array2string - imported by numpy.core (delayed), numpy.testing._private.utils (delayed)
missing module named numpy.lib.imag - imported by numpy.lib (delayed), numpy.testing._private.utils (delayed)
missing module named numpy.lib.real - imported by numpy.lib (delayed), numpy.testing._private.utils (delayed)
missing module named numpy.lib.iscomplexobj - imported by numpy.lib (delayed), numpy.testing._private.utils (delayed)
[... cut ...]
missing module named numpy.dtype - imported by numpy (top-level), numpy._typing._array_like (top-level), numpy.array_api._typing (top-level), numpy.ma.mrecords (top-level), numpy.ctypeslib (top-level)
missing module named numpy.ndarray - imported by numpy (top-level), numpy._typing._array_like (top-level), numpy.ma.core (top-level), numpy.ma.extras (top-level), numpy.ma.mrecords (top-level), numpy.ctypeslib (top-level)
missing module named numpy.ufunc - imported by numpy (top-level), numpy._typing (top-level)
missing module named numpy.histogramdd - imported by numpy (delayed), numpy.lib.twodim_base (delayed)
missing module named httplib - imported by Utils (delayed)

macfifa, 2023-05-09:
i find the problem 
pyinstaller donot recognize import done by exec(f"import {name}") in sender.py and toolpage.py so i list the import classes 
one by one and it works

engrjawadulhassan, 2023-05-16:
Can you please share the working "bCNC.spec" file here ?

engrjawadulhassan, 2023-05-21:
I successfully build an executable for the official latest github version for windows x64.
Here is the link as github only allowed me to upload 25 MB.
`https://www.upload.ee/files/15253609/exe.win-amd64-3.11.zip.html`

Harvie, 2023-05-21:
Can you please describe how you did the build?
Anyway. i would strongly reccomend for anyone to use pip install instead.
Are there any benefits of using .exe build? I can mostly think of downsides (eg. lack of infrastructure for easy upgrades,...)

engrjawadulhassan, 2023-05-22:
Hello,

I used PyCharm to debug and remove the script and build errors (Deprecated commands (e.g., stdout instead of print, etc.) and dependencies errors). Then compiled in one.

.exe has many benefits, for Windows users who don't want to mess with Python installation on their systems (especially for newbies).
Anyone can run directly a compiled .exe file without installing Python or other related things on their systems.
You can run it from a USB flash drive using any laptop or PC anywhere.

So I think team bCNC should go for .exe compilation for future releases.

I hope you will understand my point.

Thanks

macfifa, 2023-07-07:
> Can you please describe how you did the build? Anyway. i would strongly reccomend for anyone to use pip install instead. Are there any benefits of using .exe build? I can mostly think of downsides (eg. lack of infrastructure for easy upgrades,...)

i want to build my own version where i have changed the DrageKnife code to write gcode for cnc (using servo for z M3 sXX for up and M3 s0 for down) to act like papercutter