vlachoudis/bCNC #1854: can not update

Baeckchen, 2023-10-03:
Hello,
I try to update my bcnc version (now 0.9.14-dev) on a rasperry pi 3 but it run into an error.
What can I do ?

`pi@Fraese:~ $ pip install --upgrade bCNC
Looking in indexes: https://pypi.org/simple, https://www.piwheels.org/simple
Collecting bCNC
  Using cached https://files.pythonhosted.org/packages/2d/be/4a49fa0b2e89ccaad0da8bd14f5dffe492c9de6b1c8c4524b0ca1234bf5b/bCNC-0.9.15.tar.gz
    Complete output from command python setup.py egg_info:
    Traceback (most recent call last):
      File "<string>", line 1, in <module>
      File "/tmp/pip-install-AwXvtE/bCNC/setup.py", line 4, in <module>
        from setuptools import find_namespace_packages, setup
    ImportError: cannot import name find_namespace_packages
        ----------------------------------------
Command "python setup.py egg_info" failed with error code 1 in /tmp/pip-install-AwXvtE/bCNC/
`

ailrst, 2023-10-21:
Have you tried updating pip? 

`pip install --upgrade pip`

Baeckchen, 2023-10-21:
I think i´ve an old Python version :-(

`pi@Fraese:~ $ pip install --upgrade pip
DEPRECATION: Python 2.7 reached the end of its life on January 1st, 2020. Please upgrade your Python as Python 2.7 is no longer maintained. pip 21.0 will drop support for Python 2.7 in January 2021. More details about Python 2 support in pip can be found at https://pip.pypa.io/en/latest/development/release-process/#python-2-support pip 21.0 will remove support for this functionality.
Defaulting to user installation because normal site-packages is not writeable
Looking in indexes: https://pypi.org/simple, https://www.piwheels.org/simple
Requirement already up-to-date: pip in ./.local/lib/python2.7/site-packages (20.3.4)
pi@Fraese:~ $ pip install --upgrade bCNC
DEPRECATION: Python 2.7 reached the end of its life on January 1st, 2020. Please upgrade your Python as Python 2.7 is no longer maintained. pip 21.0 will drop support for Python 2.7 in January 2021. More details about Python 2 support in pip can be found at https://pip.pypa.io/en/latest/development/release-process/#python-2-support pip 21.0 will remove support for this functionality.
Defaulting to user installation because normal site-packages is not writeable
Looking in indexes: https://pypi.org/simple, https://www.piwheels.org/simple
Collecting bCNC
  Using cached bCNC-0.9.15.tar.gz (1.3 MB)
    ERROR: Command errored out with exit status 1:
     command: /usr/bin/python -c 'import sys, setuptools, tokenize; sys.argv[0] = '"'"'/tmp/pip-install-hCgI0m/bcnc/setup.py'"'"'; __file__='"'"'/tmp/pip-install-hCgI0m/bcnc/setup.py'"'"';f=getattr(tokenize, '"'"'open'"'"', open)(__file__);code=f.read().replace('"'"'\r\n'"'"', '"'"'\n'"'"');f.close();exec(compile(code, __file__, '"'"'exec'"'"'))' egg_info --egg-base /tmp/pip-pip-egg-info-le_eJW
         cwd: /tmp/pip-install-hCgI0m/bcnc/
    Complete output (5 lines):
    Traceback (most recent call last):
      File "<string>", line 1, in <module>
      File "/tmp/pip-install-hCgI0m/bcnc/setup.py", line 4, in <module>
        from setuptools import find_namespace_packages, setup
    ImportError: cannot import name find_namespace_packages
    ----------------------------------------
ERROR: Command errored out with exit status 1: python setup.py egg_info Check the logs for full command output.`

ailrst, 2023-10-23:
I fixed this by re-installing the latest version of Raspbian so I could install python3 and install bcnc with `pip3`, but if your raspbian isn't too old you may just be able to install a newer Python.

Baeckchen, 2023-10-23:
Thanks for your help, I´ll think I have to upgrade my hard- and software :-(