vlachoudis/bCNC #1872: Large files issue random straight lines

massiveradiation, 2024-01-13:
### Discussed in https://github.com/vlachoudis/bCNC/discussions/1871

<div type='discussions-op-text'>

<sup>Originally posted by **massiveradiation** January 13, 2024</sup>
Hey folks, 
I've been using bCNC for years now on an RPi and it has been awesome! Seriously, I love this software and I want to thank everyone involved in its development. 

Over the years I've been having one persistent issue with random straight line movement in large files. I've ruled out Fusion 360 and post processing for grbl. I've ruled out the hardware. It boils down to bCNC randomly inserting straight line movements. 

[image]

You can see in the image, below and parallel to the green line, how the machine simply stopped executing gcode and moved in a straight random line, and then continued with the gcode as if nothing happened. I repeat, this is not in the gcode.  I've analyzed and plotted the gcode at ncviewer.com (if you haven't used this, you're missing out!) and it's perfect. Sometimes several of these random straight lines are cut over the course of the job.  The file size is almost 15MB, almost 600K lines of gcode. It's simply a repeating pattern, so I wish there were a way to simply repeat the same bit of gcode over and over, but from different start points. In this case, it would reduce the number of gcode lines to less than 7K! Anyway, a man can dream.

I'm using the latest Beta 9.14 version. Perhaps I should step back to 9.11?

I know that there hasn't been any movement on the bCNC github for quite some time now, but I'm hoping someone may have some insight since I love this software and want to continue using it, but damage is occurring randomly to my work pieces. The autolevel feature is something I can't live without for my purposes. I've looked at other possible gcode senders, but none have as easy and repeatable autolevel setup as bCNC. 

I'm going to try breaking the large gcode files into pieces to see if that will have any effect. This is not ideal, as I can't just walk away and let this job run in its entirety, but perhaps it's better than losing work, time and material. 

Any input or suggestions to help solve this perplexing riddle, are welcome. 

Cheers,
Tyler 

</div>

MARIOBASZ, 2024-01-14:
If you are using automatic level and use G53 in some movement, that may be the cause of ruining the material.

massiveradiation, 2024-01-14:
Interesting! I'll take a look. I'm not sure, but I'm using a pattern folder in Fusion 360 and you would think that all of the instances of the exact same pattern would be the same. But, I'll definitely do a search for G53 in the gcode just in case. 
Thanks for your input!

massiveradiation, 2024-01-14:
I checked for G53 in the file, none exist.