vlachoudis/bCNC #1885: gcode is displayed in a small scrollable frame.

Git-Bruno, 2024-04-28:
The frame showing the gcode is a small scrollable frame that cannot be enlarged to occupy the entire frame at the left of the bCNC window. This is with the last master branch, code dated Apr,22,2024. 
[image]

Harvie, 2024-04-29:
Thanks for noticing this. i did not. wonder which commit had introduced it...

Git-Bruno, 2024-04-30:
This was introduced on Fev 09 by PR#1849, Vertical scrollbar to make small screens easier to use. bmain.py was changed to use VScrollFrame.py.
If I replace bmain.py with a copy not using VScrollFrame.py then the gcode fills the entire left frame.
I tested previous versions and the left frame in the Editor page was already scrollable by default.
To me the code introduced by PR#1849 should either not apply when in the Editor page or , for all pages, always adapt the size of the left frame to the screen size and make it scrollable when the content cannot fit in the frame.

ownedcore, 2024-07-08:
Observing the same with a Debian 12 installation. Also cant figure out which version of bCNC i actually have installed. Though I have the VScrollFrame.py file.

If I change line 39 of VScrollFrame.py and add ", height=event.height" the frame gets all down to the bottom (Editor tab). Also this renders the ability to scroll down, if the left side frame is not occupying the whole window. Tested on "File" tab.

Other thing i notice without the change to VScrollFrame.py and the height addition is that i can scroll the left side up when its not fully occupying available space.

Im not familiar with tk so cant help, just wanted to share my obversation.

Git-Bruno, 2024-07-29:
This update to VScrollFrame.py woks for me too. May be this change is ready to roll if the display on small screen works as per code change for PR#1849.

Harvie, 2024-09-24:
Fixed by reverting

closed 2024-09-24 by Harvie