vlachoudis/bCNC #1888: Slice mesh doesn't work

HolyBlackCat, 2024-05-27:
When trying to use the "slice mesh" plugin, I get this error:
```
AttributeError: module 'numpy' has no attribute 'int'.
`np.int` was a deprecated alias for the builtin `int`. To avoid this error in existing code, use `int` by itself. Doing this will not modify any behavior and is safe. When replacing `np.int`, you may wish to use e.g. `np.int64` or `np.int32` to specify the precision. If you wish to review your current use, check the release note link for additional information.
```

Harvie, 2024-05-27:
Thanks for pointing this out. Can you please prepare PR? I will merge it.

HolyBlackCat, 2024-05-28:
Sure, https://github.com/vlachoudis/bCNC/pull/1889

closed 2024-05-28 by Harvie