vlachoudis/bCNC #1907: After upgrade from LM21.3 to 22

ltspicer, 2024-10-11:
$ ./bCNC.sh
Launching bCNC from .
================================================================================
WARNING: bCNC was recently ported to only support 
python3.8 and newer.
Most things seem to work reasonably well.
Please report any issues to: https://github.com/vlachoudis/bCNC/issues
================================================================================
new-config bCNC <configparser.ConfigParser object at 0x75840ef81880>
Traceback (most recent call last):
  File "<frozen runpy>", line 198, in _run_module_as_main
  File "<frozen runpy>", line 88, in _run_code
  File "/home/daniel/bCNC/bCNC/__main__.py", line 226, in <module>
    main()
  File "/home/daniel/bCNC/bCNC/__main__.py", line 46, in main
    import bmain
  File "/home/daniel/bCNC/bCNC/bmain.py", line 62, in <module>
    import CNCCanvas
  File "/home/daniel/bCNC/bCNC/CNCCanvas.py", line 51, in <module>
    from CNC import CNC
  File "/home/daniel/bCNC/bCNC/CNC.py", line 25, in <module>
    from svgcode import SVGcode
  File "/home/daniel/bCNC/bCNC/lib/svgcode.py", line 14, in <module>
    from svgelements import SVG, Arc, Close, Line, Move, Path, Shape
ModuleNotFoundError: No module named 'svgelements'

I use Python2 too (for Oki scan tool)
python ist Python 2 and python3 is Python 3.12.3

What can I do?

ltspicer, 2024-10-11:
Or: how I reinstall/compile...

JipeB, 2024-10-11:
Hello, the svgelements library is missing.
Just do pip install svgelements from the command line.
JP

ltspicer, 2024-10-11:
$ pip install svgelements
DEPRECATION: Python 2.7 reached the end of its life on January 1st, 2020. Please upgrade your Python as Python 2.7 is no longer maintained. pip 21.0 will drop support for Python 2.7 in January 2021. More details about Python 2 support in pip, can be found at https://pip.pypa.io/en/latest/development/release-process/#python-2-support
Defaulting to user installation because normal site-packages is not writeable
Collecting svgelements
  Downloading svgelements-1.9.6-py2.py3-none-any.whl (137 kB)
     |████████████████████████████████| 137 kB 4.4 MB/s 
Installing collected packages: svgelements
Successfully installed svgelements-1.9.6
WARNING: You are using pip version 20.1.1; however, version 20.3.4 is available.
You should consider upgrading via the '/usr/bin/python2 -m pip install --upgrade pip' command.
daniel@daniel-System-Product-Name ~ $ bCNC
Traceback (most recent call last):
  File "/home/daniel/.local/bin/bCNC", line 5, in <module>
    from bCNC.__main__ import main
ModuleNotFoundError: No module named 'bCNC'
daniel@daniel-System-Product-Name ~ $ pip install --upgrade bCNC
DEPRECATION: Python 2.7 reached the end of its life on January 1st, 2020. Please upgrade your Python as Python 2.7 is no longer maintained. pip 21.0 will drop support for Python 2.7 in January 2021. More details about Python 2 support in pip, can be found at https://pip.pypa.io/en/latest/development/release-process/#python-2-support
Defaulting to user installation because normal site-packages is not writeable
Collecting bCNC
  Downloading bCNC-0.9.15.tar.gz (1.3 MB)
     |████████████████████████████████| 1.3 MB 4.1 MB/s 
    ERROR: Command errored out with exit status 1:
     command: /usr/bin/python2 -c 'import sys, setuptools, tokenize; sys.argv[0] = '"'"'/tmp/pip-install-N6SFeq/bCNC/setup.py'"'"'; __file__='"'"'/tmp/pip-install-N6SFeq/bCNC/setup.py'"'"';f=getattr(tokenize, '"'"'open'"'"', open)(__file__);code=f.read().replace('"'"'\r\n'"'"', '"'"'\n'"'"');f.close();exec(compile(code, __file__, '"'"'exec'"'"'))' egg_info --egg-base /tmp/pip-pip-egg-info-URydM2
         cwd: /tmp/pip-install-N6SFeq/bCNC/
    Complete output (5 lines):
    Traceback (most recent call last):
      File "<string>", line 1, in <module>
      File "/tmp/pip-install-N6SFeq/bCNC/setup.py", line 4, in <module>
        from setuptools import find_namespace_packages, setup
    ImportError: cannot import name find_namespace_packages
    ----------------------------------------
ERROR: Command errored out with exit status 1: python setup.py egg_info Check the logs for full command output.
WARNING: You are using pip version 20.1.1; however, version 20.3.4 is available.
You should consider upgrading via the '/usr/bin/python2 -m pip install --upgrade pip' command.

Harvie, 2024-10-11:
upgrade to python 3 please. the transition took years, but whole world did already moved on from python 2 now...

ltspicer, 2024-10-11:
I have installed Python3.12
But too Python2 for Oki scantool.
Is bCNC ready for Python3.12?
I coded a Linux installer for dxf2gcode ( https://github.com/ltspicer/dxf2gcode.linux.installer ).
Python3.11- is not the same than 3.12+. I had to adapt the script!

JipeB, 2024-10-11:
bCNC installation procedure with python > 3.10

The installation of bCNC in the python source folder is no longer possible since version 3.10; you must create a virtual environment. Ref: https://www.docstring.fr/glossaire/environnement-virtuel/

The procedure is done in command line in an input terminal (right click on the desktop to open the terminal). Enter the commands in bold.

1- Create a virtual environment
venv CNC (CNC as an example, any name can be chosen)

2- Activate this environment
source bin/activate

3- Load the modules needed for bCNC to work with the pip function
pip install pillow
pip install opencv-python
pip install numpy
pip install svgelements
pip install shxparser
pip install pyserial
pip install scipy

4- Check that the modules are installed in CNC/lib/python 3.1x/site-packages

5- If everything is correct, open the site-packages folder as an administrator and paste the bCNC folder contained in bCNC-master into it.

To launch bCNC, create a launcher and search for the directory of the installed python version. Add _m bCNC to the end of the command string.
/office/CNC/lib/python3.1x -m bCNC (x varies depending on the installed version.
 Best regards,
 JP

Harvie, 2024-10-11:
> 3- Load the modules needed for bCNC to work with the pip function
> pip install pillow
> pip install opencv-python
> pip install numpy
> pip install svgelements
> pip install shxparser
> pip install pyserial
> pip install scipy

wait a moment... pip should install those automaticaly when you install bCNC

JipeB, 2024-10-11:
With python3.12 and the obligation to use a virtual environment it does not work:

cncpi@raspberrypi:~ $ pip3 install --upgrade bCNC
error: externally-managed-environment

× This environment is externally managed
╰─> To install Python packages system-wide, try apt install
python3-xyz, where xyz is the package you are trying to
install.

If you wish to install a non-Debian-packaged Python package,
create a virtual environment using python3 -m venv path supprimer le lien
Then use path/to/venv/bin/python and path supprimer le lien Make
sure you have python3-full installed.

pip install bCNC tries to install in the python root and throws an error.
With my procedure it works, but there is surely a way to automate the process.

I faced the problem recently when ubuntu upgraded from 22.04 to 24.04 with python 3.12 integrated. My system which had been running for 3 years was no longer working. I had to do the manual installation again to restart my machine.
JP

Harvie, 2024-10-11:
> With python3.12 and the obligation to use a virtual environment it does not work

then obviously it should work by installing bCNC within that venv.

OR

if you're naughty boy like me, then you can just remove `/usr/lib/python3.12/EXTERNALLY-MANAGED` file :-)

JipeB, 2024-10-12:
It is actually much simpler.

ltspicer, 2024-10-12:
Thanks @ all
I have now downgraded to LM21.3

Harvie, 2024-10-12:
No idea what is LM tho...

ltspicer, 2024-10-12:
> No idea what is LM tho...

Linux Mint ;)