vlachoudis/bCNC #1908: Ubuntu 24.04.1LTS - Installation procedure changed

Lubensius, 2024-10-13:
I updated to Ubuntu 24.04.1LTS and bCNC was nor longer staring I had to install it by
`
rm .local/bin/bCNC #to prevent error in later installation
sudo apt install python3-full #maybe not needed
sudo apt install pipx    #this is now mandatory
pipx install bCNC
`
maybe README should be updated

rrffggee, 2024-11-14:
A great thanks for you,  this saved my install of bCNC.
Works with Ubuntu 22.04 too 
rrffggee

jmomadit, 2024-12-26:
works almost fine for me (24.04.1LTS), but I got "module compiled against ABI version 0x1000009 but this version of numpy is 0x2000000" after calling bCNC. The program runs well, but the camera won't work.

downgrading numpy did the job for me (seems like opencv had a problem with the installed version):
pipx inject bCNC numpy==1.26.4

gmartinezo, 2025-02-01:
> works almost fine for me (24.04.1LTS), but I got "module compiled against ABI version 0x1000009 but this version of numpy is 0x2000000" after calling bCNC. The program runs well, but the camera [...]

Work's for me Ubuntu 22.04

roboticboyer, 2025-04-30:
In case is used UV and not PIPX
`uvx --with numpy==1.26.4 bCNC`