vlachoudis/bCNC #1910: Can't run bCNC on Windows 11 with Python 3.13

Neil8978, 2024-11-10:
I installed Python 3.13 from Python.org on a new Windows 11 notebook, then installed bCNC using 'pip install --upgrade bCNC' as per instructions from 'https://pypi.org/project/bCNC/' The following results when I attempt to run bCNC:
c:\Users\Neil>python -m bCNC
================================================================================
WARNING: bCNC was recently ported to only support
python3.8 and newer.
Most things seem to work reasonably well.
Please report any issues to: https://github.com/vlachoudis/bCNC/issues
================================================================================
new-config bCNC <configparser.ConfigParser object at 0x00000210E1D797F0>
RuntimeError: module compiled against ABI version 0x1000009 but this version of numpy is 0x2000000
Traceback (most recent call last):
  File "<frozen runpy>", line 198, in _run_module_as_main
  File "<frozen runpy>", line 88, in _run_code
  File "C:\Users\Neil\AppData\Roaming\Python\Python313\site-packages\bCNC\__main__.py", line 226, in <module>
    main()
    ~~~~^^
  File "C:\Users\Neil\AppData\Roaming\Python\Python313\site-packages\bCNC\__main__.py", line 153, in main
    application = bmain.Application(className=f"  {Utils.__prg__}  ")
  File "C:\Users\Neil\AppData\Roaming\Python\Python313\site-packages\bCNC\bmain.py", line 147, in __init__
    Sender.__init__(self)
    ~~~~~~~~~~~~~~~^^^^^^
  File "C:\Users\Neil\AppData\Roaming\Python\Python313\site-packages\bCNC\Sender.py", line 91, in __init__
    self.controllerLoad()
    ~~~~~~~~~~~~~~~~~~~^^
  File "C:\Users\Neil\AppData\Roaming\Python\Python313\site-packages\bCNC\Sender.py", line 134, in controllerLoad
    self.controllers[name] = eval(f"{name}.Controller(self)")
                             ~~~~^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "<string>", line 1, in <module>
NameError: name 'G2Core' is not defined. Did you mean: 'GCode'?

c:\Users\Neil>

I could not see any previous reports of this error but there are some similarities to issue #1901

jpagliaccio, 2024-11-10:
I have the same problem on Linux (Fedora 41) solved it by also installing and running with Python 3.9. 

Here is the error from Python 3.11. Looks the same as Windows. Hope it helps. Please let me know if you want me to test anything. 

Thanks so much for bCNC.  

```
(bcnc) bluenuc:bCNC: python3.13 bCNC.py
This is currently broken. Use instead: python -m bCNC
bCNC runpy loader: /opt/bcnc_master/bCNC/bCNC/__main__.py
================================================================================
WARNING: bCNC was recently ported to only support 
python3.8 and newer.
Most things seem to work reasonably well.
Please report any issues to: https://github.com/vlachoudis/bCNC/issues
================================================================================
new-config bCNC <configparser.ConfigParser object at 0x7fda4df46660>
Traceback (most recent call last):
  File "/opt/bcnc_master/bCNC/bCNC.py", line 20, in <module>
    runpy.run_path(bcncpath, run_name="__main__")
    ~~~~~~~~~~~~~~^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "<frozen runpy>", line 287, in run_path
  File "<frozen runpy>", line 98, in _run_module_code
  File "<frozen runpy>", line 88, in _run_code
  File "/opt/bcnc_master/bCNC/bCNC/__main__.py", line 226, in <module>
    main()
    ~~~~^^
  File "/opt/bcnc_master/bCNC/bCNC/__main__.py", line 153, in main
    application = bmain.Application(className=f"  {Utils.__prg__}  ")
  File "/opt/bcnc_master/bCNC/bCNC/bmain.py", line 147, in __init__
    Sender.__init__(self)
    ~~~~~~~~~~~~~~~^^^^^^
  File "/opt/bcnc_master/bCNC/bCNC/Sender.py", line 91, in __init__
    self.controllerLoad()
    ~~~~~~~~~~~~~~~~~~~^^
  File "/opt/bcnc_master/bCNC/bCNC/Sender.py", line 134, in controllerLoad
    self.controllers[name] = eval(f"{name}.Controller(self)")
                             ~~~~^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "<string>", line 1, in <module>
NameError: name 'G2Core' is not defined. Did you mean: 'GCode'?
(bcnc) bluenuc:bCNC: 

```

Neil8978, 2024-11-11:
Many thanks for getting back to me so quickly. I will revert to Python 3.9 which worked on my old now deceased notebook. Again thanks and regards Neil Sent from my Galaxy

Neil8978, 2024-11-11:
HiI removed Python 3.13 and installed 3.9, bCNC is now running although I had to upgrade the opencv-python to get the camera working.Many thanks for your help. Regards Neil Sent from my Galaxy

silver2row, 2025-02-11:
This is on WSL2...

1. `sudo apt install python3-venv`
2. `mkdir SOME_DIR && cd SOME_DIR`
3. `python3 -m venv .` 
4. `source bin/activate`
5. `python -m pip install --upgrade bCNC`

While in your virtual environment in WSL2, just type `bCNC` and it should run and pop up.

Also, if you are using WSL2 under a debian OS, you can just use this command to get it installed:

```
sudo apt install bcnc
```

Lylesgold, 2025-07-24:
This seems to be the same issue as reported in #1947. I attempted to get Python 3.9.22 on my Win 11 PC, but how to do that is not straightforward. If this is a persistent issue, could someone please produce a new exe for bCNC.  Thanks.