vlachoudis/bCNC #1917: bCNC is very slow on Rpi4

jdelahayes, 2024-12-01:
I've been working with bCNC on an Rpi3 for a while without any problems. I would like to install bCNC on a Rpi4. The last version of raspberrypi OS is a Debian 12.8 with Python 3.11.2 and I reinstalled bCNC without problem from the git repository.

But it is very slow, the tabs are displayed gradually and when you load a gcode it becomes unusable.
Has anyone encountered the same kind of problem on a recent installation of bCNC on Rpi4?

jdelahayes, 2024-12-01:
After testing and researching, the slowness is probably due to using xWayland since Debain Bookworm.

Similar problem found on Thonny project:
https://github.com/thonny/thonny/issues/3036#issuecomment-1872394923

I downgrade to Debian bullseye. bCNC UI performance is as usual.

WheeSci, 2024-12-14:
Having the same problem.  Mouse moves too slow to be used.

rottaran, 2025-06-30:
I can confirm the performance problems on a Raspberry Pi 5 with latest Raspberry Pi OS. The sluggishness seems to increase with the complexity of the job and rendering of the window. 

Following blog post from the KiCAD project pointed into the direction of OpenGL call throttling caused by wayland: https://www.kicad.org/blog/2025/06/KiCad-and-Wayland-Support/ Indeed, the performance problems where resolved by switching from wayland to X11 through raspi-config.

rottaran, 2025-07-01:
Oh, didn't expect that. Still the performance problem was resolved by switching from wayland to X11.

Now that I think about it, the whole GUI was rendering slowly with wayland. I mean that all the buttons appeared individually slow enough on screen to be noticable. With the X11 system it is not noticable.

Harvie, 2025-07-01:
bCNC is purely X11 application, on wayland it has to use Xwayland "emulation" layer, which obviously can introduce some extra latency...

fbonin5491, 2025-11-28:
confirmed , after disabling wayland through raspi-config , no more problems.same with owon-tiny oscilloscope that became usable.