vlachoudis/bCNC #1918: "\n"  needs to be replaced by   b"\n"   in line 268 @ bCNC/Pendant.py

ldpgh, 2025-01-01:
```
#####  OS  ###################################
$ uname -a
Linux Linux-Latitude-E5570 5.15.0-122-generic #132-Ubuntu SMP Thu Aug 29 13:45:52 UTC 2024 x86_64 x86_64 x86_64 GNU/Linux
```

```
##### bCNC  ################################
$ python3 -m pip show bCNC
Name: bCNC
Version: 0.9.15
Summary: Swiss army knife for all your CNC/g-code needs
Home-page: https://github.com/vlachoudis/bCNC
Author: Vasilis Vlachoudis
Author-email: vvlachoudis@gmail.com
License: GPLv2
Location: /home/hhggff/.local/lib/python3.10/site-packages
Requires: numpy, opencv-python, Pillow, pyserial, shxparser, svgelements
Required-by: 
```

```
#####  error-message/-log  ######################
$ python3 -m bCNC
================================================================================
WARNING: bCNC was recently ported to only support 
python3.8 and newer.
Most things seem to work reasonably well.
Please report any issues to: https://github.com/vlachoudis/bCNC/issues
================================================================================
new-config bCNC <configparser.ConfigParser object at 0x7f6fdc0e27d0>
G2Core loaded
/usr/lib/python3/dist-packages/scipy/__init__.py:146: UserWarning: A NumPy version >=1.17.3 and <1.25.0 is required for this version of SciPy (detected version 1.26.4
  warnings.warn(f"A NumPy version >={np_minversion} and <{np_maxversion}"
self.canvas.winfo_id(): 67108952
Not Loading 6 axis displays
Not Loading 6 axis displays
192.168.178.21 - - [01/Jan/2025 21:32:28] "192.168.178.21" : "GET / HTTP/1.1" 200 -
192.168.178.21 - - [01/Jan/2025 21:32:28] "192.168.178.21" : "GET /send?gcode=%24%23%0A%24G%0A HTTP/1.1" 200 -
----------------------------------------
Exception occurred during processing of request from ('192.168.178.21', 34238)
Traceback (most recent call last):
  File "/home/hhggff/.local/lib/python3.10/site-packages/bCNC/Pendant.py", line 264, in mainPage
    f = open(os.path.join(webpath, page), "rb")
FileNotFoundError: [Errno 2] No such file or directory: '/home/hhggff/.local/lib/python3.10/site-packages/bCNC/pendant/favicon.ico'

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "/usr/lib/python3.10/socketserver.py", line 316, in _handle_request_noblock
    self.process_request(request, client_address)
  File "/usr/lib/python3.10/socketserver.py", line 347, in process_request
    self.finish_request(request, client_address)
  File "/usr/lib/python3.10/socketserver.py", line 360, in finish_request
    self.RequestHandlerClass(request, client_address, self)
  File "/usr/lib/python3.10/socketserver.py", line 747, in __init__
    self.handle()
  File "/usr/lib/python3.10/http/server.py", line 433, in handle
    self.handle_one_request()
  File "/usr/lib/python3.10/http/server.py", line 421, in handle_one_request
    method()
  File "/home/hhggff/.local/lib/python3.10/site-packages/bCNC/Pendant.py", line 180, in do_GET
    self.mainPage(page[1:])
  File "/home/hhggff/.local/lib/python3.10/site-packages/bCNC/Pendant.py", line 268, in mainPage
    self.wfile.write("\n".join([
TypeError: sequence item 0: expected str instance, bytes found
----------------------------------------
```

```
#####  fix  ######################################################
replace 
    self.wfile.write("\n".join([
by
    self.wfile.write(b"\n".join([
```

```
#####  check permutions of fix (string vs binary)  #######################
$ python3
Python 3.10.12 (main, Sep 11 2024, 15:47:36) [GCC 11.4.0] on linux
Type "help", "copyright", "credits" or "license" for more information.
>>>
>>> "\n".join(["aa","bb"])
'aa\nbb'
>>>
>>> b"\n".join(["aa","bb"])
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
TypeError: sequence item 0: expected a bytes-like object, str found
>>>
>>> b"\n".join([b"aa",b"bb"])
b'aa\nbb'
>>>
>>> "\n".join([b"aa",b"bb"])
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
TypeError: sequence item 0: expected str instance, bytes found
```

Harvie, 2025-01-02:
Can you please make PR, i will make sure to merge it.

ldpgh, 2025-01-04:
Issue closed by reporter ... issue has been solved already in 'master'.

My bCNC has been fresh installed by 'pip' 2 days ago.
But the package provided by 'pip' has different content than latest Github-repo.

Current Github-repo does not have this issue any longer.
BTW line number in current Github-repo is 297 and no longer 268 as reported.

closed 2025-01-04 by ldpgh

Harvie, 2025-01-04:
Looks like i should update the pip tho...

ldpgh, 2025-01-04:
Thx ... but wait with update-
There is one more issue in Pendant.py
I'm preparing the issue right now ... easy fix.

ldpgh, 2025-01-04:
Issue & PR (https://github.com/vlachoudis/bCNC/pull/1920) created.