vlachoudis/bCNC #1919: bCNC/Pendant.py issues runtime error due to 'enum' while 'str' is expected.

ldpgh, 2025-01-04:
```
#####  OS  ###################################
$ uname -a
Linux Linux-Latitude-E5570 5.15.0-122-generic #132-Ubuntu SMP Thu Aug 29 13:45:52 UTC 2024 x86_64 x86_64 x86_64 GNU/Linux
```

```
##### bCNC  ################################
$ python3 -m pip show bCNC
Name: bCNC
Version: 0.9.15   ... directly installed from Github 'master' after uninstall old version
Summary: Swiss army knife for all your CNC/g-code needs
Home-page: https://github.com/vlachoudis/bCNC
Author: Vasilis Vlachoudis
Author-email: vvlachoudis@gmail.com
License: GPLv2
Location: /home/az42/.local/lib/python3.10/site-packages
Requires: numpy, opencv-python, Pillow, pyserial, shxparser, svgelements
Required-by: 

$ python3
Python 3.10.12 (main, Nov  6 2024, 20:22:13) [GCC 11.4.0] on linux
Type "help", "copyright", "credits" or "license" for more information.
>>> 
```

```
#####  error-message/-log  ######################
$ python3 -m bCNC
================================================================================
WARNING: bCNC was recently ported to only support 
python3.8 and newer.
Most things seem to work reasonably well.
Please report any issues to: https://github.com/vlachoudis/bCNC/issues
================================================================================
new-config bCNC <configparser.ConfigParser object at 0x7fbd7182a950>
G2Core loaded
self.canvas.winfo_id(): 100663390
Not Loading 6 axis displays
Not Loading 6 axis displays
log_message ('GET /state HTTP/1.1', '200', '-') <class 'tuple'> <class 'str'>
log_message (<HTTPStatus.REQUEST_URI_TOO_LONG: 414>, 'Request-URI Too Long') <class 'tuple'> <enum 'HTTPStatus'>
----------------------------------------
Exception occurred during processing of request from ('127.0.0.1', 50158)
Traceback (most recent call last):
  File "/usr/lib/python3.10/socketserver.py", line 316, in _handle_request_noblock
    self.process_request(request, client_address)
  File "/usr/lib/python3.10/socketserver.py", line 347, in process_request
    self.finish_request(request, client_address)
  File "/usr/lib/python3.10/socketserver.py", line 360, in finish_request
    self.RequestHandlerClass(request, client_address, self)
  File "/usr/lib/python3.10/socketserver.py", line 747, in __init__
    self.handle()
  File "/usr/lib/python3.10/http/server.py", line 433, in handle
    self.handle_one_request()
  File "/usr/lib/python3.10/http/server.py", line 406, in handle_one_request
    self.send_error(HTTPStatus.REQUEST_URI_TOO_LONG)
  File "/usr/lib/python3.10/http/server.py", line 463, in send_error
    self.log_error("code %d, message %s", code, message)
  File "/usr/lib/python3.10/http/server.py", line 565, in log_error
    self.log_message(format, *args)
  File "/home/kk/.local/lib/python3.10/site-packages/bCNC/Pendant.py", line 59, in log_message
    if strargs0.startswith("GET / ") or strargs0.startswith("GET /send"):
AttributeError: 'HTTPStatus' object has no attribute 'startswith'
----------------------------------------
log_message (<HTTPStatus.REQUEST_URI_TOO_LONG: 414>, 'Request-URI Too Long') <class 'tuple'> <enum 'HTTPStatus'>
----------------------------------------
Exception occurred during processing of request from ('127.0.0.1', 50164)
Traceback (most recent call last):
  File "/usr/lib/python3.10/socketserver.py", line 316, in _handle_request_noblock
    self.process_request(request, client_address)
. . . . .

Issue is '<enum 'HTTPStatus'>' is given as args[0] to log_message() at line 54.
The 'if' statement in line 56 tries to handle it as string  (args[0].startswith...)
It fails in case of args[0] gets an enum.

Why log_message() gets
   *args=(<HTTPStatus.REQUEST_URI_TOO_LONG: 414>, 'Request-URI Too Long')
is not clear to me.
```

```
#####  fix  ######################################################
surround args[0] with str()
```

Harvie, 2025-01-04:
Can you please make PR?

ldpgh, 2025-01-04:
Issue has been solved by PR https://github.com/vlachoudis/bCNC/pull/1920.
PR accepted and added to 'master'-branch.

closed 2025-01-04 by ldpgh