vlachoudis/bCNC #1931: Milling with two tools of different diameters (useful for PCBs)

MARIOBASZ, 2025-03-21:
My idea for a milling suite with two tools of different diameters, starting from the original path (or the smaller cutter pass), seems to be working correctly:
For the larger diameter cutter:
1- Repeat the idea of ​​not entering adjacent paths (island milling).
For the smaller diameter cutter:
1- Mill the areas where the larger diameter cutter can't reach.
2- Mill the corners and angles where the larger diameter cutter can't reach (this option should be made optional, as it is quite time-consuming).

MARIOBASZ, 2025-03-21:
Blue bit 0.7mm
Black bit 0.10mm

[image]

MARIOBASZ, 2025-03-21:
[image]

lalo-uy, 2025-03-22:
And how do you generate the second path data ?

MARIOBASZ, 2025-03-22:
It's a plugin. The file with the tracks (either actual size or small-diameter cutters) is loaded. The "small bit, gaps only" option is being fixed.