vlachoudis/bCNC #1934: grbl version detection issue and strange "Stream requested by CYCLE START machine button" loop

g3grau, 2025-04-13:
I'm trying to get my Genmitsu3030 running with bCNC 0.9.16 (linux py3.12.9). In the bCNC console it tells me
```
Grbl 1.1f ['$' for help]
Grbl for ARM32
Version:ARM32 V3.0

```
which seems to break the version detection due to two lines of code being sent, one having "for" after Grbl. In controllers/_GenericController.py around line 298 apparently a string "GRBL0" or "GRBL1" should be assembled with
```
            if self.master.controller in ("GRBL0", "GRBL1"):
                    self.master.controllerSet(
                        "GRBL%d" % (int(CNC.vars["version"][0])))

```
but the 2nd line causes an exception trying to convert "f" to int. I wrote an ugly fix like this
```
            if self.master.controller in ("GRBL0", "GRBL1"):
                if CNC.vars["version"][0][0] in ["0", "1"]:
                    self.master.controllerSet(
                        "GRBL" + CNC.vars["version"][0][0])

```
which stops the error but still doesn't let me use bCNC. The program now loops like
```
Controller state changed to: Idle (Running: True)
Ignoring machine stream request, because of state:  Idle True
Ignoring machine stream request, because of state:  Idle True
Ignoring machine stream request, because of state:  Idle True
Ignoring machine stream request, because of state:  Idle True
Ignoring machine stream request, because of state:  Idle True
Ignoring machine stream request, because of state:  Idle True
Ignoring machine stream request, because of state:  Idle True
Controller state changed to: Idle (Running: False)
Job done. Purging the controller. (Running: False)
Stream requested by CYCLE START machine button
Will clean after this operation
```
which also repeats the version check. I guess this error may be due to not properly fixing the issue (and not by some machine button as the message tells). Any ideas how to fix this? The Machine works fine with Candle2 or openBuilds control, thus I rule out hardware or connection issues.

R4id0ss, 2025-06-09:
Same issue, any update?

Harvie, 2025-06-09:
>  yeah I can set it, but gives a long stream of data like this:

maybe try without `?raw` it should be more readable... im not sure

Harvie, 2025-06-09:
nevermind, i've decoded it... it seems your firmware is actualy sending the start signal: and couple more for that matter:

`<Idle|MPos:0.000,0.000,0.000|Bf:35,254|FS:0,0|Pn:RHS>`

important part is 

`Pn:RHS`

Harvie, 2025-06-09:
> Genmitsu3030

Is that another chinese macine? they tend to have their grbl setup quite broken, because they don't really care about firmware and just wanna sell the hardware... should probably build and flash your own grbl to be sure it's OK. but i reckon it's some non-standard HW, so might be bit tricky...

R4id0ss, 2025-06-09:
just for reply:

<le|MPos:0.000,0.000,0.000|Bf:3254|FS:0,0|Pn:RHS>
<e|MPos:0.000,0.000,0.000|Bf:34|FS:0,0|Pn:RHS>
<e|MPos:0.000,0.000,0.000|Bf:3FS:0,0|Pn:RHS>
<MPos:0.000,0.000,0.000|Bf:35FS:0,0|Pn:RHS>
<MPos:0.000,0.000,0.000|Bf:3FS:0,0|Pn:RHS>
<IMPos:0.000,0.000,0.000|Bf:35,254|FS:0,0|Pn:RHS>

Grbl 1.1f ['$' for help]

Grbl for ARM32
Version:ARM32 V3.0
$0=10 (step pulse,usec)
$1=255 (step idle delay,msec)
$2=0 (stepport invert mask)
$3=3 (dirport invert mask)
$4=0 (stepenable invert,bool)
$5=1 (lims pin invert,bool)
$6=1 (probe pin invert,bool)
$10=3 (status report mask)
$11=0.010 (junction deviation)
$12=0.002 (arc tolerance,mm)
$13=0 (report inches,bool)
$20=0 (soft limits,bool)
$21=1 (hard limits,bool)
$22=1 (home cycle,bool)
$23=3 (homing dir invert mask)
$24=50.000 (homing feed,mm/min)
$25=500.000 (homing seek,mm/min)
$26=250 (homing debounce,msec)
$27=1.000 (homing pull-off,mm)
$30=12000 (maximum spindle speed,rpm)
$31=0 (minimum spindle speed,rpm)
$32=0 (laser mode enable,bool)
$100=787.402 (X axis pulse:step/mm)
$101=800.000 (Y axis pulse:step/mm)
$102=800.000 (Z axis pulse:step/mm)
$110=2000.000 (X axis max rata:mm/min)
$111=2000.000 (Y axis max rata:mm/min)
$112=2000.000 (Z axis max rata:mm/min)
$120=30.000 (X axis acceleration:mm/s^2)
$121=30.000 (Y axis acceleration:mm/s^2)
$122=30.000 (Z axis acceleration:mm/s^2)
$130=200.000 (X aixs max travel:mm)
$131=200.000 (Y aixs max travel:mm)
$132=80.000 (Z aixs max travel:mm)
kk

R4id0ss, 2025-06-09:
> nevermind, i've decoded it... it seems your firmware is actualy sending the start signal: and couple more for that matter:
> 
> `<Idle|MPos:0.000,0.000,0.000|Bf:35,254|FS:0,0|Pn:RHS>`
> 
> important part is
> 
> `Pn:RHS`
> 
> From grbl docs:
> 
> Input Pin State:
> 
> ```
> Pn:XYZPDHRS indicates which input pins Grbl has detected as 'triggered'.
> 
> Pin state is evaluated every time a status report is generated. All input pin inversions are appropriately applied to determine 'triggered' states.
> 
> Each letter of XYZPDHRS denotes a particular 'triggered' input pin.
>     X Y Z XYZ limit pins, respectively
>     P the probe pin.
>     D H R S the door, hold, soft-reset, and cycle-start pins, respectively.
>     Example: Pn:PZ indicates the probe and z-limit pins are 'triggered'.
>     Note: A may be added in later versions for an A-axis limit pin.
> ```
OK, understood.

I can not understand why it doesn't work, with Candle or UGS it works flawlessly.

Can I do something to fix this?
thank you

R4id0ss, 2025-06-09:
> Is that another chinese macine? they tend to have their grbl setup quite broken, because they don't really care about firmware and just wanna sell the hardware... should probably build and flash [...]

the same, 3030 Prover.
Thank you for the explain.

There is any guide for that?

R4id0ss, 2025-06-09:
https://github.com/user-attachments/assets/2729fbd0-571e-478a-8552-d9a6f7c9f2f1

[image]

This is the board and its chip.

I'll try to contact the vendor and ask.

I would like to use bCNC for his orientation gcode amd maybe other chinese cnc users can benefit from it.

Harvie, 2025-06-09:
I was looking up some more info about that genmitsu 3030 prover max and it looks very impressive. Seems like it does hit the sweetspot for hobby desktop cnc. I am now almost tempted to be geting one 😅 but i dont think i should.

But you should definitely get in touch with manufacturer to get clear details about the board and firmware.

R4id0ss, 2025-06-09:
> Why does it have VGA connector? Is there video output of some kind?

It is not a VGA conn. 
It's an MPG connector.

R4id0ss, 2025-06-09:
> I was looking up some more info about that genmitsu 3030 prover max and it looks very impressive. Seems like it does hit the sweetspot for hobby desktop cnc. I am now almost tempted to be geting one [...]

Yeah, i really good machine.

I'll update this issue with vendor reply.