vlachoudis/bCNC #1935: Is it possible to use a file with non-grid pattern for autolevel feature?

muhammadelmogy, 2025-04-18:
I want to know how does the grid points generated in the Autolevel feature and is it possible that those points are not in a grid pattern?

I am working on a tool to do auto level for PCB and have the ability to avoid drill points so the generated grid points will not exactly in the a grid pattern. something like this:
[image]

As you can see some points will be shifted from its original place to avoid probing a drill point and other points may be omitted totally from the grid like the points around the drill points at the corners, you can click on the image and zoom in for more details 

So my plan is to use those generated points and create the height map and save the result in a file to use it in BCNC!
Is it possible to use a file with non-grid pattern for Autolevel feature?

massiveradiation, 2025-04-18:
There are no abilities in bCNC to change the probing pattern. Out of
curiosity, why not use the CNC to drill the holes?

lalo-uy, 2025-04-19:
> There are no abilities in bCNC to change the probing pattern. Out of <br>
> curiosity, why not use the CNC to drill the holes? <br> <br> On Fri, Apr
> <br> <br> &gt; I want to know how does the grid points generated in the
> Autolevel feature <br> &gt; and is it possible that those points are not in
> a grid pattern? <br> &gt; <br> &gt; I am working on a tool to do auto level
> for PCB and have the ability to <br> &gt; avoid drill points so the
> generated grid points will not exactly in the a <br> &gt; grid pattern.
> &gt; &lt;
> <br> &gt; <br> &gt; As you can see some points will be shifted from its
> original place to <br> &gt; avoid probing a drill point and other points
> may be omitted totally from <br> &gt; the grid like the points around the
> drill points at the corners, you can <br> &gt; click on the image and zoom
> in for more details <br> &gt; <br> &gt; So my plan is to use those
> generated points and create the height map and <br> &gt; save the result in
> a file to use it in BCNC! <br> &gt; Is it possible to use a file with
> non-grid pattern for Autolevel feature? <br> &gt; <br> &gt; — <br> &gt;
> this thread.Message <br> &gt; ID: ***@***.***&gt; <br> &gt;
> https://github.com/vlachoudis/bCNC/issues/1935&gt; <br> &gt; <br> &gt; I
> <br> &gt; and is it possible that those points are not in a grid pattern?
> <br> &gt; <br> &gt; I am working on a tool to do auto level for PCB and
> have the ability to <br> &gt; avoid drill points so the generated grid
> points will not exactly in the a <br> &gt; grid pattern. something like
> this thread.Message <br> &gt; ID: ***@***.***&gt; <br> &gt; <br>

If you made the pcb with Plated Thru Holes (PTH) the holes are drilled with
the CNC, then the board is plated and only then it is engraved. So if you
mathe the autolevel before engraving the holes are already there.

The autolevel values are used to interpolate the z value at a point in
between.
You will have to llok at the code if it uses the fixed distant between grid
points or each point coordinates to do that. In the second case , if the
point if a little off the grid all may work the same.

muhammadelmogy, 2025-04-19:
The height map should only be created after plating the holes and to do so I must avoid the drill points when creating the map or else the map will fail.
if the a height map was created before drilling the holes then it would be useless since the board has been removed for plating!

I really do want to open the code and look inside it because I am not very expert in python and the code is not simple and it will take so much time to reach the required results so if one of the developers can help with the answer then it will save a huge time and effort for me

MARIOBASZ, 2025-05-29:
I'm not discovering anything new: drilling with a diameter larger than the holes can serve as an alternative solution.