vlachoudis/bCNC #1947: Wrong numpy version

Curtalain, 2025-06-25:
Bonjour
I have this error when  I run python -m bCNC 

> new-config bCNC <configparser.ConfigParser object at 0x10c945af0>
> RuntimeError: module compiled against ABI version 0x1000009 but this version of numpy is 0x2000000
> macOS 13 (1307) or later required, have instead 13 (1306) !
> zsh: abort      python -m bCNC

Can you help me

Curtalain, 2025-06-26:
> How did you installed bCNC? Are you running on macOS?

I install bCNC with homebrew

I am running on macOs Ventura 13.7.6

Bien à toi

*Sincèrement*

*Alain CURTET*

chrisherman, 2025-07-03:
pip uninstall numpy
pip install 'numpy<2'

Harvie, 2025-07-03:
> pip uninstall numpy
> pip install 'numpy<2'

i don't get what the problem is... was there some api change in numpy? maybe we should fix it in code to be compatible with some reasonably recent version.